Omniflux/esphome-fujitsu-halcyon #1: Fujitsu indoor unit temperature

icare3416, 2023-12-12:
Hello,
Thank you very much for your work.
I am in equivalent situation  9a page 6 of https://hvacdirect.com/hvac/pdf/fj-rc-wifi-1na-installation-guide.pdf.
I have made a electronic circuit with ESP32 and Lin component to communicate to CN6 (Fujitsu Indoor Unit)

1) I have tried
`controller_address: 0  # 0=Primary, 1=Secondary`
`temperature_controller_address: 0  # Fujitsu controller address to read temperature from`
and
`controller_address: 0  # 0=Primary, 1=Secondary`
`temperature_controller_address: 1  # Fujitsu controller address to read temperature from`
I can stop/start the unit ... But I can't see Fujitsu indoor unit temperature.
Do you have a idea ?

2) example.yaml line 56
For me, I have to change
`- platform: esphome_fujitsu_halcyon`
by 
`- platform: fujitsu_halcyon`

Best regards

Omniflux, 2023-12-12:
If you have a Fujitsu, Intesis, or additional ESP remote controller + ESP, the ESP can report the temperature from the other remote controller.

The indoor unit does not expose its temperature reading, only the remote controllers do - in 9a and 9b, the blue arrow shows where the temperature reading can be read from by the ESP.

Thank you for reporting esphome_fujitsu_halcyon vs fujitsu_halcyon.

icare3416, 2023-12-12:
Your answer is clear.
Thank you for your help.

closed 2023-12-12 by Omniflux