Omniflux/esphome-fujitsu-halcyon #2: Fujitsu miminum heat

jfdrolet11, 2025-03-11:
Thanks for the good work! I installed it 1 week ago and everything is fine except that I can no longer set the minimum heat mode (heat @ 10 degrees Celsius) from the IR remote controller. I understand that the mode is not currently supported in the component, but I would have thought that I could have been able to continue to use this mode when I am away. When I press the MIN. HEAT button on the IR remote controller, the heat pump emits a beep and the operation indicator lamp flashes. There is no change in the ESPHome climate thermostat and it seems to ignore the command received. Is it possible that the IR remote controller can set the minimum heat mode even if the component doesn’t support it yet?

Omniflux, 2025-03-11:
I unfortunately don't have a unit with IR that I can connect to (mine appear to require an addon communications board).

Did this work previously with https://github.com/FujiHeatPump/esphome-fujitsu or is this your first integration with the unit?

Do you know the model number of the indoor unit?

jfdrolet11, 2025-03-11:
It's the first integration I've tried. I found your integration easier to configure. I'll try the other integration and see. 

My indoor unit is ASU18RLF.

Thanks!

Omniflux, 2025-03-12:
I cannot find it in any of the manuals, but in this marketing brochure https://www.washingtonenergy.com/assets/uploads/HFI_Multi_Brochure.pdf it states "Minimum Heat, Sleep Timer functions will be lost when the wired remote controller is installed"

I am working on another component for the central remote control for the outdoor unit if yours supports a central remote control. The central controller does not disable MIN. HEAT, but I don't have that portion done yet, and no time frame.

johnjohn-11, 2025-03-12:
Great find! Thanks for your help. I'll follow your repo for the new integration. No hurry for me. I can wait as long as it takes!

In the meantime, I'll make myself a Home Assistant automation and alternate between heat and fan mode to simulate the minimum heat feature.

johnjohn-11, 2025-04-08:
I can close this feature request now. In the end, the feature is not available when it has a wired thermostat connected to the indoor unit.

closed 2025-04-08 by johnjohn-11