Omniflux/esphome-fujitsu-halcyon #3: question and issue

iotEFFIX, 2025-03-24:
Hello,

i just had a couple of questions. So if i am right i don't have to implement any of the .cpp or .h files into esphome to work because we all get them extern via:
 external_components:
  - source: github://Omniflux/esphome-fujitsu-halcyon
  - source: github://Omniflux/esphome-tzsp
 right?
My issue if this is correct, and i just have to copy and paste the yaml file and change some names where needed.
How to find the file:
packages:
  wifi: !include common/wifi.yaml
 because this file can't be found and is not delivered in this repository i think.

Omniflux, 2025-03-24:
Correct, external_components now takes care of the .cpp, and .h files.

Packages are explained here https://esphome.io/components/packages.html

If you are not using packages, just replace that section with the wifi or ethernet section from your existing yaml.

You probably also need to change the pins set in the uart section. The example yaml showing tx 13 rx 14 is probably not correct for your device.

If you are using an Atom Lite + FOSV's [Fuji-Atom-Interface](https://github.com/FOSV/Fuji-Atom-Interface) it would be

uart:
  tx_pin: GPIO22
  rx_pin: GPIO19

The circuit in Aaron Zhang's original post I believe uses

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16

iotEFFIX, 2025-03-25:
Thank you for the answer! Now i have working yaml file that is downloadable. I can also upload it to my esp32 feather and it confirms is has been installed correctly. I just can't get life into it for some reason, no climate template and no acces to the local acces to the site of the esphome project. Do i miss something or is something wrong?
Here is my .yaml file:
```
substitutions:
  name: airco-jd-test
  friendly_name: Airco Jürgen & Davy
  halcyon_name: aircoj&d
  esp_board: featheresp32

external_components:
  - source: github://Omniflux/esphome-fujitsu-halcyon
  - source: github://Omniflux/esphome-tzsp

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: ${esp_board}
  framework:
    type: esp-idf

api:
  password: ""

web_server:
  port: 80

ota:
  platform: esphome
  password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Fujitsu"
    password: !secret hotspot_password

sensor:
- platform: uptime
  name: "Uptime"

button:
- platform: restart
  name: Restart device
- platform: safe_mode
  name: Safe Mode Boot

#logger:
#  level: DEBUG

uart:
  tx_pin: GPIO17  # Device dependent
  rx_pin: GPIO16  # Device dependent
  baud_rate: 500
  parity: EVEN

climate:
- platform: fujitsu-halcyon
  name: ${halcyon_name}
  # Fujitsu devices use 0 and 1, but 2-15 should also work. Must not skip addresses
  controller_address: 1  # 0=Primary, 1=Secondary
  #temperature_controller_address: 0  # Fujitsu controller address to read temperature from
  #temperature_sensor_id: my_temp_sensor  # ESPHome sensor to read temperature from
  #ignore_lock: true  # Ignore feature lock

  # To capture communications for debugging / analysis
  # Use Wireshark with https://github.com/Omniflux/fujitsu-airstage-h-dissector
#  tzsp:
#    ip: 192.168.1.20
#    protocol: 255
```

Omniflux, 2025-03-25:
Nothing looks obviously wrong with your configuration.

I would remove the external_component, uart, and climate sections and see if local access works.
If it does, I would suspect uart pins are incorrect. If not, esp_board or wifi settings.

iotEFFIX, 2025-03-26:
Yes, when i delete the things you said here above i can enter the local site with the buttons that are in the yaml file still. The GPIO pins should be correct since i came here to your project because of Aaron Zhang's. I checked this myself to. When i upload the yaml file like i posted above, i get the following error in repeating mode:

`[09:21:24]ets Jul 29 2019 12:21:46
[09:21:24]
[09:21:24]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:21:24]configsip: 0, SPIWP:0xee
[09:21:24]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:21:24]mode:DIO, clock div:1
[09:21:24]load:0x3fff0030,len:7112
[09:21:24]load:0x40078000,len:15624
[09:21:24]load:0x40080400,len:4
[09:21:24]load:0x40080404,len:3876
[09:21:24]entry 0x4008064c
[09:21:24]I (58) boot: ESP-IDF 5.1.5 2nd stage bootloader
[09:21:24]I (58) boot: compile time Mar 26 2025 09:07:43
[09:21:24]I (58) boot: Multicore bootloader
[09:21:24]I (63) boot: chip revision: v3.0
[09:21:24]I (67) boot.esp32: SPI Speed      : 80MHz
[09:21:24]I (71) boot.esp32: SPI Mode       : DIO
[09:21:24]I (76) boot.esp32: SPI Flash Size : 4MB
[09:21:24]W (80) boot.esp32: PRO CPU has been reset by WDT.
[09:21:24]W (86) boot.esp32: WDT reset info: PRO CPU PC=0x400fa096
[09:21:24]W (92) boot.esp32: WDT reset info: APP CPU PC=0x400f96d6
[09:21:24]I (98) boot: Enabling RNG early entropy source...
[09:21:24]I (103) boot: Partition Table:
[09:21:24]I (107) boot: ## Label            Usage          Type ST Offset   Length
[09:21:24]I (114) boot:  0 otadata          OTA data         01 00 00009000 00002000
[09:21:24]I (122) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[09:21:24]I (129) boot:  2 app0             OTA app          00 10 00010000 001c0000
[09:21:24]I (137) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[09:21:24]I (144) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[09:21:24]I (152) boot: End of partition table
[09:21:24]I (156) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=35568h (218472) map
[09:21:24]I (231) esp_image: segment 1: paddr=00045590 vaddr=3ffb0000 size=03bcch ( 15308) load
[09:21:24]I (236) esp_image: segment 2: paddr=00049164 vaddr=40080000 size=06eb4h ( 28340) load
[09:21:24]I (247) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=ceba4h (846756) map
[09:21:24]I (502) esp_image: segment 4: paddr=0011ebcc vaddr=40086eb4 size=0ec4ch ( 60492) load
[09:21:24]I (535) boot: Loaded app from partition at offset 0x10000
[09:21:24]I (535) boot: Disabling RNG early entropy source...
[09:21:24]I (546) cpu_start: Multicore app
[09:21:24]I (547) cpu_start: Pro cpu up.
[09:21:24]I (547) cpu_start: Starting app cpu, entry point is 0x40081ccc
[09:21:24]I (0) cpu_start: App cpu up.
[09:21:24]I (564) cpu_start: Pro cpu start user code
[09:21:24]I (564) cpu_start: cpu freq: 160000000 Hz
[09:21:24]I (564) cpu_start: Application information:
[09:21:24]I (569) cpu_start: Project name:     airco-jd-test
[09:21:24]I (575) cpu_start: App version:      2025.3.2
[09:21:24]I (580) cpu_start: Compile time:     Mar 26 2025 09:06:24
[09:21:24]I (586) cpu_start: ELF file SHA256:  2e1b6607afc0532b...
[09:21:24]I (592) cpu_start: ESP-IDF:          5.1.5
[09:21:24]I (597) cpu_start: Min chip rev:     v0.0
[09:21:24]I (601) cpu_start: Max chip rev:     v3.99 
[09:21:24]I (606) cpu_start: Chip rev:         v3.0
[09:21:24]I (611) heap_init: Initializing. RAM available for dynamic allocation:
[09:21:24]I (618) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[09:21:24]I (624) heap_init: At 3FFB9220 len 00026DE0 (155 KiB): DRAM
[09:21:24]I (630) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[09:21:24]I (637) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[09:21:24]I (643) heap_init: At 40095B00 len 0000A500 (41 KiB): IRAM
[09:21:24]I (650) spi_flash: detected chip: generic
[09:21:24]I (654) spi_flash: flash io: dio
[09:21:24]I (659) app_start: Starting scheduler on CPU0
[09:21:24]I (663) app_start: Starting scheduler on CPU1
[09:21:24]I (663) main_task: Started on CPU0
[09:21:24]I (673) main_task: Calling app_main()
[09:21:24]I (827) main_task: Returned from app_main()`

Omniflux, 2025-03-27:
Could you set your logger level to `debug` and post about 10 seconds of the log?

It should contain lines like these, among other things

```
[20:44:21][D][fujitsu-halcyon:071][UART_Event]: RX: 00 A0 00 0B 16 A0 03 20
[20:44:22][D][fujitsu-halcyon:071][UART_Event]: RX: 20 A1 00 0B 16 00 42 00
[20:44:22][D][fujitsu-halcyon:071][UART_Event]: TX: 21 81 00 0B 16 20 00 00
[20:44:22][D][fujitsu_halcyon_controller::Controller:096][UART_Event]: UART break!
```

Then set `controller_address` to `2` and post about 10 seconds of the log again. If set to 2, it should remain in read only mode, and not put your primary controller into an error state.

It would be great if the log covered the period when the Fujitsu unit first powers on, but should be helpful even if you can't get that.

iotEFFIX, 2025-03-31:
Hey, sorry for the late answer. I only work on the project on 3 days a week. when i set it to debug it is giving me this:
`INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/aircojd2.yaml...
INFO Starting log output from [redacted] using esphome API
INFO Successfully connected to [redacted] in 0.064s
INFO Successful handshake with [redacted] in 0.012s
[09:41:41][I][app:100]: ESPHome version 2025.3.2 compiled on Mar 31 2025, 09:36:22
[09:41:41][C][wifi:600]: WiFi:
[09:41:41][C][wifi:428]:   Local MAC: [redacted]
[09:41:41][C][wifi:433]:   SSID: [redacted]
[09:41:41][C][wifi:436]:   IP Address: [redacted]
[09:41:41][C][wifi:439]:   BSSID: [redacted]
[09:41:41][C][wifi:441]:   Hostname: [redacted]
[09:41:41][C][wifi:443]:   Signal strength: -74 dB ▂▄▆█
[09:41:41][C][wifi:447]:   Channel: 1
[09:41:41][C][wifi:448]:   Subnet: [redacted]
[09:41:41][C][wifi:449]:   Gateway: [redacted]
[09:41:41][C][wifi:450]:   DNS1: [redacted]
[09:41:41][C][wifi:451]:   DNS2: [redacted]
[09:41:41][C][logger:177]: Logger:
[09:41:41][C][logger:178]:   Max Level: DEBUG
[09:41:41][C][logger:179]:   Initial Level: DEBUG
[09:41:41][C][logger:181]:   Log Baud Rate: 115200
[09:41:41][C][logger:182]:   Hardware UART: UART0
[09:41:41][C][uart.idf:159]: UART Bus 1:
[09:41:41][C][uart.idf:160]:   TX Pin: GPIO17
[09:41:41][C][uart.idf:161]:   RX Pin: GPIO16
[09:41:41][C][uart.idf:163]:   RX Buffer Size: 256
[09:41:41][C][uart.idf:165]:   Baud Rate: 500 baud
[09:41:41][C][uart.idf:166]:   Data Bits: 8
[09:41:41][C][uart.idf:167]:   Parity: EVEN
[09:41:41][C][uart.idf:168]:   Stop bits: 1
[09:41:41][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[09:41:41][C][uptime.sensor:033]:   Device Class: 'duration'
[09:41:41][C][uptime.sensor:033]:   State Class: 'total_increasing'
[09:41:41][C][uptime.sensor:033]:   Unit of Measurement: 's'
[09:41:41][C][uptime.sensor:033]:   Accuracy Decimals: 0
[09:41:41][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[09:41:41][C][uptime.sensor:034]:   Type: Seconds
[09:41:41][C][restart.button:017]: Restart Button 'Restart Device'
[09:41:41][C][restart.button:017]:   Icon: 'mdi:restart'
[09:41:41][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
[09:41:41][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[09:41:41][C][fujitsu-halcyon:075]: FujitsuHalcyonController:
[09:41:41][C][fujitsu-halcyon:076]:   Controller Address: 1 (Secondary)
[09:41:41][C][fujitsu-halcyon:077]:   Remote Temperature Controller Address: 0 (Primary)
[09:41:41][C][fujitsu-halcyon:078]:   Remote Temperature Controller Sensor 'Remote Temperature Sensor'
[09:41:41][C][fujitsu-halcyon:078]:     Device Class: 'temperature'
[09:41:41][C][fujitsu-halcyon:078]:     State Class: 'measurement'
[09:41:41][C][fujitsu-halcyon:078]:     Unit of Measurement: '°C'
[09:41:41][C][fujitsu-halcyon:078]:     Accuracy Decimals: 0
[09:41:41][C][fujitsu-halcyon:080]:   Use Sensor: NO
[09:41:41][C][fujitsu-halcyon:081]:   Ignore Lock: NO
[09:41:41][C][web_server:285]: Web Server:
[09:41:41][C][web_server:286]:   Address: [redacted]
[09:41:41][C][mdns:116]: mDNS:
[09:41:41][C][mdns:117]:   Hostname: [redacted]
[09:41:41][C][esphome.ota:073]: Over-The-Air updates:
[09:41:41][C][esphome.ota:074]:   Address: [redacted]
[09:41:41][C][esphome.ota:075]:   Version: 2
[09:41:41][C][esphome.ota:078]:   Password configured
[09:41:41][C][safe_mode:018]: Safe Mode:
[09:41:41][C][safe_mode:019]:   Boot considered successful after 60 seconds
[09:41:41][C][safe_mode:021]:   Invoke after 10 boot attempts
[09:41:41][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[09:41:41][C][api:140]: API Server:
[09:41:41][C][api:141]:   Address: [redacted]
[09:41:41][C][api:145]:   Using noise encryption: NO
[09:41:43][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[09:41:43][D][esp32.preferences:114]: Saving 1 preferences to flash...
[09:41:43][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[09:42:25][D][sensor:093]: 'Uptime': Sending state 102.23700 s with 0 decimals of accuracy`
I redacted every sensetive information obviously.

When i set the controller adress to 2 i get the same.

Omniflux, 2025-04-01:
This does not show it receiving any data from the Fujitsu unit, so probably uart pins or something is disconnected. Is this a Feather V2? It looks like the V2 uses pins 7(rx) and 8(tx). There are a lot of Feather models and many of them use different pins...

iotEFFIX, 2025-04-01:
I use the adafruit esp32 feather.

Omniflux, 2025-04-02:
Not sure where to go from here, there should have been a UART_Event in the log if anything had been received on the serial port, even bad data.