Omniflux/esphome-fujitsu-halcyon #5: Question about   temperature_sensor_id: my_temp_sensor

tevaadarelsol, 2025-04-02:
Hi,

First of all, thank you very much for your work and effort. This project addresses some of the needs of those of us who have a Fujitsu air conditioner and want to be able to integrate it with Home Assistant.

I'm using your code and it works perfectly. However, I have a question regarding the line:

#temperature_sensor_id: my_temp_sensor # ESPHome sensor to read temperature from

What is this parameter for? Based on the comment, I assume it's used to tell the air conditioning unit the outside temperature, for example, from a separate thermostat integrated into Home Assistant? If so, how would I pass that value to ESPHome?

My air conditioning unit covers four rooms, but I'm only interested in cooling the room I'm in. So I'd like a way to tell it which thermostat to look at, depending on which room I'm in. However, it's not clear to me how I should send that value to it or how ESPHome should collect it.

Am I correct? I hope I've explained myself well enough. English isn't my native language, and I'm using a translator.

Thank you again for this amazing project.

Omniflux, 2025-04-02:
The Fujitsu indoor units have a temperature sensor in their air intake which they usually use for deciding when and what mode to operate in. Some remote controls such as the UTY-RNNUM have a temperature sensor built in which can be used to override this sensor.

Using a remote sensor like this requires both enabling its use on the remote control and enabling its use on the indoor unit by programming a feature code. If your unit supports this, instructions will be in your installation manual (not the operating manual). Look for a section called "Function setting". On my unit, it is function settings `42` and `48`.

<br />

If you have a Fujitsu remote with this sensor as your primary controller and your esphome device is secondary, setting `temperature_controller_address: 0` will report this value to Home Assistant as the current temperature for this climate device.

If your esphome device is your primary controller and your Fujitsu remote is secondary, setting `temperature_controller_address: 1` will report this value to Home Assistant as the current temperature for this climate device.

| [image] | 
|:--:| 
| Temperature reported to Home Assistant (left) vs default (right) |

<br />

Instead of using the Fujitsu remote sensor, you can use an esphome sensor. In this case, set `temperature_sensor_id: my_temp_sensor` where `my_temp_sensor` is defined elsewhere in your yaml file.

If your esphome device does not have its own sensor, you can pull one from Home Assistant like this

```
sensor:
  - platform: homeassistant
    id: my_temp_sensor
    entity_id: sensor.my_temp_sensor # This is the Home Assistant entity_id
```

If you do this, it will report this value to Home Assistant as the current temperature for this climate device (I suspect this only works if your sensor is in Celsius right now).

<br />

IF your indoor unit supports using a remote sensor AND you have set the function setting to enable it AND configured a sensor in your yaml file using one of the methods described above, you should now have a switch named `Use Sensor` in the Home Assistant device page under the Configuration section. Enabling this switch will begin reporting this temperature to the indoor unit.

<br />
Some Fujitsu remote controls allow locking some features (child lock, part lock, etc.). `ignore_lock: true` will ignore these when set, as they are enforced on the remote control and not the indoor unit.

tevaadarelsol, 2025-04-04:
Thank you so much for clarifying the configuration parameters.

My indoor unit supports the use of a remote sensor. In fact, the original remote has a built-in thermistor, and the temperature is retrieved from it (I can verify this by breathing on the thermostat and seeing that the temperature recorded in Home Assistant changes).

However, I've configured the ESPHome module with:

controller_address: "1"
temperature_controller_address: "1"

and

sensor:
- platform: homeassistant
id: my_temp_sensor
entity_id: <my_external_temperature_sensor_entity_id>

But I cannot see the "Use sensor" switch in the device settings in the ESPHome integration, and therefore, I can't get the temperature to be retrieved from my external thermostat.

Do you have any idea what I might be doing wrong?

Thank you so much for your help!

Omniflux, 2025-04-04:
You should be using `temperature_sensor_id: my_temp_sensor`, not `temperature_controller_address: 1`

```
sensor:
  - platform: homeassistant
    id: my_temp_sensor
    entity_id: ...

climate:
  - platform: fujitsu-halcyon
  - name: "..."
  - controller_address: 1
  - temperature_sensor_id: my_temp_sensor
```

<br/>
<br/>

If this is what you are doing and the posted configuration is incorrect, then I can think of three possibilities

1. Sensor switching is not enabled on the unit
2. The unit does not report the sensor switching feature the same way mine do
3. There is a timing issue in the code, and Home Assistant is picking up the traits before the unit finishes reporting its features

If it is the last issue, you can force the switch to appear

```
sensor:
  - platform: homeassistant
    id: my_temp_sensor
    entity_id: ...

climate:
  - platform: fujitsu-halcyon
  - name: "..."
  - controller_address: 1
  - temperature_sensor_id: my_temp_sensor
  - use_sensor:
      name: "Use Sensor"
      internal: false
```

but if it is the first issue, this will produce a switch that appears to work, but actually does nothing.

tevaadarelsol, 2025-04-05:
> You should be using `temperature_sensor_id: my_temp_sensor`, not `temperature_controller_address: 1`

You are the man!

Indeed, this was the problem. I mistakenly thought I should specify which controller the temperature was being taken from and, from that controller, specify which external thermostat to use.

Leaving only "temperature_sensor_id: my_temp_sensor" is able to read the temperature from the external entity correctly, although the "Use sensor" switch doesn't appear on the device configuration page. It's possible to force it to appear with the configuration you indicate, but I don't mind since it doesn't affect operation.

By the way, what is the "Reinitialize" button for? Is it to restart the ESP32? Is there a way to hide it?

Thanks again for your help.

Omniflux, 2025-04-05:
Reinitialize restarts the communication state, not the whole device. Pressing it will cause the device to re-request the features list from the indoor unit, and search for another remote control at the next controller_address. For example, if your esp device is your primary controller and you connect a Fujitsu remote control as secondary, you can use this button instead of power cycling the indoor unit.

You should not need it in normal operation. You can disable or change the visibility of the button in the Home Assistant device configuration page, like anything else.

[image]

I am not sure about on the esp device web_server page, you could try

```
climate:
  - platform: fujitsu-halcyon
  - name: "..."
  - controller_address: 1
  - reinitialize:
      name: "Reinitialize"
      internal: true
      disabled_by_default: true
```
for this, but I have not verified if it works.

closed 2025-12-10 by tevaadarelsol