Omniflux/esphome-fujitsu-halcyon #6: Comportamiento extraño en dos placas identicas

tevaadarelsol, 2025-04-04:
Hi mate,

Thank you again for your effort and your work bringing us this wonderful component.

I wanted to ask you a question: I have two identical air conditioning units at home, for which I built two boards using ESP32 and your code to control them.

One of them works perfectly; however, the other, despite the identical code (I've only changed the name for the climate component), doesn't work, and I only receive these messages on the console again and again:

ontroller:096] [1;31m[UART_Event] [0;36m: UART break!
RX: 00 A0 10 00 00 00 00 00
RX: 20 A1 00 22 00 84 29 00
TX: 21 81 20 00 00 00 00 00
ontroller:096] [1;31m[UART_Event] [0;36m: UART break!
RX: 00 A0 10 00 00 00 00 00
RX: 20 A1 00 22 00 84 29 00
TX: 21 81 20 00 00 00 00 00
ontroller:096] [1;31m[UART_Event] [0;36m: UART break!
RX: 00 A0 10 00 00 00 00 00
RX: 20 A1 00 22 00 84 29 00
TX: 21 81 20 00 00 00 00 00

As I said, the two electronic circuits are identical, with identical components and identical setup.

Do you have any clue as to what might be going on?

Thank you very much for your help!

Omniflux, 2025-04-04:
What does doesn't work mean?
Does the primary remote work but you cannot control with the esp device?
If so, does the component report the current settings in Home Assistant?

Or does the primary remote stop working as well?

<br/>

Assuming you have `controller_address: 1` in your yaml file, the log you posted appears normal and shows the esp device is recognized by the primary remote and has copied its settings.

tevaadarelsol, 2025-04-05:
Sorry, you are absolutely correct. "It doesn't work" is too abstract to understand.

I'll try to explain myself better. The unit that works properly does indeed work correctly: the remote control works fine, and through ESPHome I can also view the settings and control the air conditioning. This is its card in Home Assistant:

[image]

[image]

However, the unit that doesn't work properly behaves strangely: although the original remote control works correctly, the ESPHome unit is able to read the current temperature but it has no control over the unit; it only has an "OFF" mode with no way to control the air conditioning:

[image]

[image]

[image]

In the logs, after the initial boot-up, WiFi connection, etc., I can only see the previous "UART Break!" messages looping, and nothing else.

I hope I've explained myself better this time. If you have any questions or need me to upload more information, please don't hesitate to let me know.

Thank you again for your help and dedication.

tevaadarelsol, 2025-04-16:
> Is there anything different about how this device is powered that would cause it to finish booting later than the working one?

Hi mate,

The two boards are identical. In fact, to rule out any possible problems, I tried swapping the ESP32 boards from one device to the other to see if the problem was with the transceiver or the ESP, but I was unsuccessful: the board that was working continues to work, and the board that wasn't working continues to not.

I've also tried other boards of a similar model (ESP32-WROOM-32) and other ESP32-S3 models, without success.

Following your comments, I've tried powering the ESP32 with an external power supply so as not to depend on the indoor unit's power supply, so that the ESP32 would start before the climate unit, but this also failed.

Thank you very much!

tevaadarelsol, 2025-04-16:
Just for some additional information, I came to this project from rabbit-aaron's:
https://community.home-assistant.io/t/fujitsu-ac-heat-pump-integration-via-esphome-esp32/407610

And with that project, it worked correctly. However, with Aaron's project, it's not possible to use an external temperature sensor, which is why I'm really interested in yours.

I'm just mentioning it in case it might help you see what's going on, knowing that the previous code works.

Thank you again!

tevaadarelsol, 2025-07-23:
Some little more information after further investigation: it seems that the configuration line it doesn't like is:

baud_rate: 500

If I compile the script with the full UART definition and a baud rate of 9600, the device boots correctly:

uart:
tx_pin: GPIO22 # Device dependent
rx_pin: GPIO19 # Device dependent
baud_rate: 9600
parity: EVEN

(Note that I'm only testing to boot the M5Stack Atom device without the FOSV module and without connecting it to the air conditioning unit)

tevaadarelsol, 2025-07-23:
More details: it only accepts baud rates of 4800 or higher. If I try anything below this, it does not work.

Omniflux, 2025-09-20:
Your second log shows the issue is a watchdog issue

```
  [09:59:47]W (85) boot.esp32: PRO CPU has been reset by WDT.
  [09:59:47]W (91) boot.esp32: WDT reset info: PRO CPU PC=0x400f371d
  [09:59:47]W (97) boot.esp32: WDT reset info: APP CPU PC=0x400f2cd2
```

I used the second configuration from your last reply, and could not reproduce the issue; the only differences I can see are that all my ATOM Lite devices are chip revision 1.0 vs your 1.1, and I am not using encryption on my API connection.

closed 2025-12-10 by tevaadarelsol