Omniflux/esphome-fujitsu-halcyon #8: Poor accuracy of the Fujitsu integrated control thermostat

danirebollo, 2025-04-15:
I've done some tests (ECO=off and use_sensor=true) with different offsets to the reported sensor temperature (the sensor used by controller and sended to unit) and seems to work bad and in a non predictable way. I mean, I want to put 20C on my room, but air doesn't power on because we're at 21, or is too slow reacting, or controller temperature switch between two values constantly even when the sensor value on Home Assistant remains the same. 

There is a good chance to optimize this using FW and some tricks like offsets to force on or off at time or it is better to go outside (like Home Assistant) and use API control to create a thermostat? Or even internally in the FW using some ESPHOME PID module (not delegating to unit)

Omniflux, 2025-04-16:
The protocol supports sensor temperature in increments of 0.5°C, but I had not coded it. I have added it in 5f02c46.

An offset can be set in the indoor units (see your indoor unit installation manual, mine is in "Remote Control Setting" -> "Function setting" -> "Temperature correction") by setting function codes. Probably 30, 31, 92, 93.

Outside of that, I think it should be handled outside of this component. The only way it can be done, I think, is by falsifying the current temperature value reported to the unit.

Omniflux, 2025-04-21:
You can fix the hassio switch like this while I investigate the correct fix.

```
climate:
  - platform: fujitsu-halcyon
    use_sensor:
      name: "Use Sensor"
      internal: false
```

danirebollo, 2025-04-22:
I've made this python GUI to sniff messages (maybe there is some bug). It uses the same TZSP connection as Wireshark, and was inspired by your LUA dissector. I found it easier than wireshark since highlights value changes.
```
# DEFINE CONFIG FILE
# current_mode: TZSP or SERIAL
# tzsp_source_ip: IP address of remote controller with TZSP protocol enabled
# tzsp_dest_ip: IP of this machine. The same configured in remote controller YAML file
# serial_port: COM port of the serial port to use in case of current_mode is SERIAL

# Config file path
CONFIG_FILE = "config.json"
DEFAULT_CONFIG = {
    "serial_port": "COM8",
    "current_mode": "TZSP",
    "tzsp_source_ip": "192.168.0.195",
    "tzsp_dest_ip": "192.168.0.167"
}

try:
    import serial
except ImportError:
    env.Execute("$PYTHONEXE -m pip install pyserial")
    import serial

try:
    import threading
except ImportError:
    env.Execute("$PYTHONEXE -m pip install threading")
    import threading

try:
    import time
except ImportError:
    env.Execute("$PYTHONEXE -m pip install time")
    import time

try:
    import tkinter as tk
except ImportError:
    env.Execute("$PYTHONEXE -m pip install tk")
    import tkinter as tk

try:
    from tkinter import ttk
except ImportError:
    env.Execute("$PYTHONEXE -m pip install tkinter")
    from tkinter import ttk

try:
    from scapy.all import *
except ImportError:
    env.Execute("$PYTHONEXE -m pip install scapy")
    from scapy.all import *

import re
import os
from datetime import datetime
import json
import sys
import subprocess
import queue

MODE_SERIAL = "SERIAL"
MODE_TZSP = "TZSP"

def load_config():
    """Load configuration from file"""
    if os.path.exists(CONFIG_FILE):
        try:
            with open(CONFIG_FILE, 'r') as f:
                return json.load(f)
[... 1492 lines cut ...]
        reading_on = False
        serial_reading = False
        actions_menu.entryconfig(0, label="Reanudar")
else:
    # TZSP mode
    reading_on = True
    serial_reading = True
    actions_menu.entryconfig(0, label="Detener")
    tzsp_handler.start()
    unrecognized_display.insert(tk.END, "TZSP capture started\n")
    unrecognized_display.see(tk.END)

# Create thread for serial reading
serial_thread = threading.Thread(target=read_serial)
serial_thread.daemon = True
serial_thread.start()

# Initialize COM port dropdown
update_com_ports()

# Start the latest counters update
root.after(100, update_latest_counters)

root.mainloop() 
```