Omniflux/esphome-fujitsu-halcyon #12: UART_HW_FIFO_LEN Compilation Error

gtmax500, 2025-07-09:
Hello, I am trying to get this installed on a m5stack Atom + FSOV board I made but am struggling to get it to compile. I have the latest version of HASS and ESPHome installed and have followed your yaml code but the compilation is failing with the following logs.

It looks like UART_FIFO_LEN was depreciated at some point in esp_idf and esp-idf 5.1 doesnt support UART_HW_FIFO_LEN but I am not sure how to use a new rev if esp-idf. Any help would be appreciated. Thanks!

```
 INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/garage-lower-ac.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing garage-lower-ac (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)

 --------------------------------------------------------------------------------

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50106.0 (5.1.6) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/garage-lower-ac/src/esphome/components/fujitsu-halcyon/Controller.cpp.o
Compiling .pioenvs/garage-lower-ac/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/garage-lower-ac/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/garage-lower-ac/src/esphome/components/logger/logger_esp32.cpp.o
src/esphome/components/fujitsu-halcyon/Controller.cpp: In member function 'bool fujitsu_halcyon_controller::Controller::start()':
src/esphome/components/fujitsu-halcyon/Controller.cpp:34:34: error: 'UART_HW_FIFO_LEN' was not declared in this scope; did you mean 'UART_FIFO_LEN'?
   34 |         const auto buffer_size = UART_HW_FIFO_LEN(this->uart_num) * 2;
      |                                  ^~~~~~~~~~~~~~~~
      |                                  UART_FIFO_LEN
*** [.pioenvs/garage-lower-ac/src/esphome/components/fujitsu-halcyon/Controller.cpp.o] Error 1
========================== [FAILED] Took 4.12 seconds ==========================
```
Here is my YAML code

```
substitutions:
  device_name: garage-lower-ac
  friendly_name: Garage Lower AC
  device_description: Atom Lite + FOSV
  esp_board: m5stack-atom

external_components:
  - source: github://Omniflux/esphome-tzsp
  - source: github://Omniflux/esphome-fujitsu-halcyon

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${device_description}

esp32:
  board: ${esp_board}
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "myapikey"
[... 246 lines cut ...]
    disabled_by_default: false
  reset_filter_timer:
    name: Reset Filter Timer
    internal: true
    disabled_by_default: false
    entity_category: config
  filter_timer_expired:
    name: Filter Timer Expired
    internal: true
    disabled_by_default: false
    entity_category: diagnostic
    device_class: problem
  reinitialize:
    name: Reinitialize
    internal: true
    disabled_by_default: false
    entity_category: config
captive_portal: {}
binary_sensor: []
number: []
switch: []
text_sensor: []

INFO Configuration is valid!
```

gtmax500, 2025-07-09:
Looking into it a bit more, I think it is because I am stuck on ESPHome 2025.5.2 which uses esp-idf 5.1 which doesnt have support for UART_HW_FIFO_LEN. I am not seeing any update available for ESPHome 2025.6 which was released June 18 but I think this is my problem. If I can get ESPHome updated to 2025.6 it should compile.

closed 2025-07-09 by gtmax500

Omniflux, 2025-07-09:
You should be able to use
```yaml
  - source: github://Omniflux/esphome-fujitsu-halcyon@f0e376a76cad39bf18878d87c5d8ba96f6d2633b
```
with ESPHome 2025.5.X