Omniflux/esphome-fujitsu-halcyon #13: Failure to change hardware to ESP32-C3

cjgarciao, 2025-07-10:
Hello,
First of all, thank you very much for your work, I am finding it very useful.
I have set up an interface based on an old ESP32 and a LIN MCP2004 transceiver and it works perfectly. As it was a bit big I decided to mount a smaller one based on an ESP32C3 mini and a MCP2025 transceiver, which doesn't work and also causes the main controller to show an error (00:00 EE). I've checked the hardware and I think it's fine. The software is the same, only changing the type of microcontroller.

Looking at the logs, I have seen that the interface that works, shows the following sequence:

```
[20:29:05][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 00 A0 00 07 9A A0 03 20
[20:29:06][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 20 A1 00 07 9A 00 37 00
[20:29:06][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: TX: 21 81 00 07 9A 20 00 00
[20:29:06][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[20:29:06][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 00 A0 00 07 9A A0 03 20
[20:29:07][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 20 A1 00 07 9A 00 37 00
[20:29:07][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: TX: 21 81 00 07 9A 20 00 00
[20:29:07][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
``` 

While the one that does not work, repeats the following sequence:
```
[20:20:53][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 00 A0 00 07 9A A0 03 20
[20:20:53][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 20 81 00 07 9A 00 37 00
[20:20:53][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 00 A0 00 07 9A A0 03 20
[20:20:54][D][esphome::fujitsu_general_airstage_h_controller:088][UART_Event]: RX: 20 81 00 07 9A 00 37 00
``` 
The difference is, on the one hand, in the reception of the second packet of the sequence (in the second byte). On the other hand, it can be seen that the one that works transmits a packet, while the one that fails does not try to transmit anything.
I think the hardware it is OK because I have tried (with another program) to transmit and with the oscilloscope I can see the signals correctly.
Any idea of what could be the problem?

Thanks,

Carlos.

cjgarciao, 2025-07-11:
Thank you for the reply and explanation.
I have put an oscilloscope on the bus line and captured data from the start, for about 15 seconds.
Globally the device is working fine, it shows sequence.

[image]

The device that doesn't work, show this secuence:

[image]

The decoding of the first three frames is (I don't know why the oscilloscope decodes it inverted with respect to the ESPHome log):

[image]

On the working device, this first sequence of three packets is exactly the same.

The next two packets:

[image]

We can see that the primary controller does not send any request to the secondary controller (it ignores it), although it must know that there is something and that's why it says “00:00 EE”.

Group 3 is two frames of the indoor unit and the response of the main controller.
Looking at the signals from the two devices, I only see a difference just at power up. The non-working device lowers the bus for a moment, but that happens long before the main controller sends its first frame (I think it might be LIN transceiver initialisation or something like that).

Do you see where the problem might be? If you need more information don't hesitate to ask me.

Carlos.

Omniflux, 2025-07-11:
I don't know why the data is transmitted inverted on the wire either, but it is :)

The first frame you decoded is a features response which looks normal, and the rest are config.

The config packets from both the primary controller and the indoor unit have the error bit set. If the first three packets are exactly the same with the working esp, it means the error bit is set there too, which is odd.

The fourth frame (first decoded config frame from the indoor unit) has the bits set for having seen both the primary and secondary controllers.

Can you decode two frames back to check if the error bit is set before either controller transmits?

One back should be `20 81 20 00 00 00 00 00` which is a feature request from the primary controller
Two back should be `00 a0 00 ...`

fourth byte, first bit will be the error bit
`00 a0 00 86 9a a0 00 20` error bit set
`00 a0 00 06 9a a0 00 20` error bit clear

cjgarciao, 2025-07-12:
I think the error bit is set because I have taken the data by switching off only the main controller and the ESP, with the indoor unit on and I guess that is why the error bit may be set. This afternoon I will take the data again with everything switched off.
Carlos.

cjgarciao, 2025-07-12:
I have taken new data by switching everything off. Now the error bit does not appear, it still does not work. I have captured about 20 seconds, the “EE” on the main controller appears a minute or two after powering on. The capture is as follows:

[image]

Frame 1: ` 00 A0 00 06 9A A0 00 20`

Frame 2: ` 00 81 00 06 9A A0 00 20`

Frame 3: ` 00 A0 00 06 9A A0 00 20`
Frame 4: ` 20 81 20 00 00 00 00 00`

Frame 5: ` 00 A0 20 1F 1F 44 00 00`
Frame 6: ` 20 A1 00 46 9A 00 37 00`
Frame 7: ` 21 A2 00 06 9A 20 00 00`

Frame 8: ` 00 A0 00 06 9A A0 03 20`
Frame 9: ` 20 81 00 06 9A 00 37 00`

Frame 10: ` 00 A0 00 06 9A A0 03 20`
Frame 11: ` 20 81 00 06 9A 00 37 00`

 If you need more information don't hesitate to ask me.

Carlos.

cjgarciao, 2025-07-13:
Hello again,

I have looked in detail at the ESP response to the first request (the group of three frames), and I can only see two differences between the one that works and the one that doesn't work.

The first difference is that in the frame sent by the main controller, the penultimate byte is different (the one that works is 0xCA and the one that doesn't works is 0xCB).

But above all, the difference that I found most important is that the not working one (ESP32C3) has a time between byte and byte of 4 ms, it is as if it were 3 STOP bits instead of one. The one that works has no additional wait.

You can see it more clearly below:

[image]

-------------------------------------------

[image]

Carlos.

fototakas, 2025-07-15:
@cjgarciao ESP32C3 works fine for me. It is not ESP32C3-Mini, but LuatOS board. What exact ESP32C3-Mini you are using?
Many ESP32C3-Mini on Aliexpress have very pure anthena design and to work in ESP-Home you need to lower transmit power:
``` yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 11 # 8.5 11 13 15 17 18.5
```

cjgarciao, 2025-07-15:
@fototakas The board is a ESP32C3 Super Mini:

[image]

I've changed the output power, but the result is the same. Thanks.

Omniflux, 2025-08-27:
I see it in the diagram, but there is nothing in the text of section 26.4.4 about it, which seems odd.

I am going to leave this issue open for now.

closed 2026-09-04 by Omniflux