Omniflux/esphome-fujitsu-halcyon #15: I can read AC status and changes but can not send commands

nicaznar, 2025-08-25:
Dear Omniflux,

 First of all, thank you very much for your work.
 I'm trying to integrate with Home Assistant an OLD oGeneral Fujitsu unit

[image]

Here is the internal unit:

oGeneral
Model: ARG25UUAN

[image]

Here is the wired remote controler:

[image]

I followed the instructions in the manual to ser the DIP pins:

 Master Unit (the wired remote controler) -> Pin 1  OFF     Pin2   OFF
 ESP32 --> Configured at Adress 2 for READING ONLY
                  "# Fujitsu devices use 0 and 1, but 2-15 should also work. Must not skip addresses
                      controller_address: 2  # 0=Primary, 1=Secondary"

With this Read Only config, the logs are like that:

[18:09:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 47 12 A0 01 20
[18:09:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 20 A1 00 47 12 00 3C 00
[18:09:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 47 12 A0 01 20
[18:09:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 20 A1 00 47 12 00 3C 00
[18:09:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 47 12 A0 01 20
[18:09:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 20 A1 00 47 12 00 3C 00
[18:09:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 47 12 A0 01 20

Only RX Activity is reported

With this config, I can read all the changes  in the master unit in HomeAssistant. I mean, If the temperature is changed in the master unit, it's reflected in Home Assistant, as well as, all the posible modes of the AC (heat, dry, only fan, etc...)

Obviously, I can not send commands to the internal unit, because ESP32 is in read only

So I really feel that I'm pretty close to achive the full control of the unit from HA

The problem comes when I tried to make the ESP32 an Slave Unit (also I tried making it master, alone and with the remote wired controler changing the DIP pins to Pin1 ON and Pin2 ON to make it slave)

When ESP32 is changed to Slave Unit , the logs are like that:

[18:11:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 10 00 00 00 00 00
[18:11:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 20 A1 00 47 12 00 3C 00
[18:11:51][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 21 81 20 00 00 00 00 00
[18:11:51][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[18:11:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 10 00 00 00 00 00
[18:11:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 20 A1 00 47 12 00 3C 00
[18:11:52][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 21 81 20 00 00 00 00 00
[18:11:52][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!

There is RX and TX activity, BUT with this config I can not send commands, NEITHER read.
BUT, If I diconnect the TX PIN form the ESP32 (with the RX pin still in place), I can read agian the status of the AC in Home Assistant

For mor info, the diagnosis in the climate component is:

Error: OK
Initialization Stage: (0/3) 

[image]

I guess, that the problem comes when the ESP32 device tries to communicate with the internal unit to advise it is part of the bus (as slave device) and it's not recognized (Am I rIght?)

There is any advise that you can give me to try to figure out how to solve it?

P.D. I also tryed to use fujitsu-airstage-h-dissector for sniffing communications, but not able to understand how it works. (may be I need more study time)
  The IP in the configuration, is the IP that the ESP32 exposes for sniffing the packets?
   #  ip: 192.168.1.20
   #  protocol: 255

Many  thanks in advance, hoping you can find time to give me and answer
Thanks

nicaznar, 2025-08-26:
Many thanks again. May I do something to help you in the investigation?
Do you need more data?

nicaznar, 2025-08-26:
It was a problem of reloading the EspHome entity. NOW I CAN CONTROL EVERYTHING!!!!

[image]

Many, many, many thanks!!!
I would like to be gratefull with you! There is something that I can do?

Omniflux, 2025-08-26:
I need to do some additional testing, but I will probably push these changes to the main branch in a week or so and you will need to change your source back to the main branch then.

I'll leave this issue open until then in case you find any remaining issues.

Omniflux, 2025-08-26:
[Climate Component - ESPHome](https://esphome.io/components/climate/)
```yaml
climate:
  - platform: ...
    visual:
      min_temperature: 18
```

nicaznar, 2025-08-27:
Many thanks again!
I finally build the circuit in a developer board, put inside a box wtih an antena and install it beside the internal unit
Working great so far!

[image]

[image]

[image]

[image]

Omniflux, 2025-09-14:
On newer units, the primary controller will keep attempting to pass the token to the secondary controller for a short period before the outdoor unit moves to an error state due to the missing secondary controller. This is usually long enough for the ESP32 to update and reboot.

If it does not work this way on older units, I am not aware of a workaround.