Omniflux/esphome-fujitsu-halcyon #18: Not working of rme

Mrsash2020, 2025-09-14:
Ok so I might be completely clueless

So my unit is a Split Fujitsu ASTG09KUCA with UTY-TWBXF1 comm unit with IR remote only

I brought the comm unit and added the Atom version. The issue is in prep of these items being received I prepped the esphome yaml and compiled it a few weeks ago. On flashing it I had all options available but the unit did not start. With the remote it beeped a few times and with esphome control in HA I could press all options. Eventually after frustration set in I did a clean build files and complied with same code and noticed I lost all options and only had the "OFF" option. So does that mean some components were removed/adjusted and why I am having issues?

[image]

esphome:
  name: ac
  friendly_name: AC
  # device_description: Atom Lite + FOSV

external_components:
  - source: github://Omniflux/esphome-tzsp
  - source: github://Omniflux/esphome-fujitsu-halcyon

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "PiEnwK4h"

captive_portal:

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "ac IP Address"
 
binary_sensor:
  - platform: status
    name: "ac Connect"
 
uart:
  tx_pin: GPIO22  # Device dependent
  rx_pin: GPIO19  # Device dependent
  baud_rate: 500
  parity: EVEN
 
climate:
  - platform: fujitsu-halcyon   # adjust to match your AC unit!
    name: "ac"
    id: climate_ac
    # Fujitsu devices use 0 and 1, but 2-15 should also work. Must not skip addresses
    controller_address: 0  # 0=Primary, 1=Secondary
    #temperature_controller_address: 0  # Fujitsu controller address to read temperature from
    temperature_sensor_id: temp
 
sensor:
  - platform: homeassistant
    name: "ac-Temp"
    entity_id: sensor.aq_temperature
    id: temp
    internal: true
 
  - platform: wifi_signal
    name: "ac WiFi Strength"
    update_interval: 60s
 
  - platform: uptime
    name:  "ac Uptime"
 
switch:
  - platform: restart
    name: "ac reboot"
  - platform: safe_mode
    name: "Restart (Safe Mode)"

Omniflux, 2025-09-14:
Is the ESP device receiving data from the UTY-TWBXF1? In the ESPHome logs, there should be lines similar to this.

```
[20:44:21][D][fujitsu-halcyon:071][UART_Event]: RX: 00 A0 00 0B 16 A0 03 20
[20:44:22][D][fujitsu-halcyon:071][UART_Event]: RX: 20 81 00 0B 16 00 42 00
[20:44:22][D][fujitsu_halcyon_controller::Controller:096][UART_Event]: UART break!
```

Mrsash2020, 2025-09-14:
ok I turned the breaker off then on and its showing signs of life

:28][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:29][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:29][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:29][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:29][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:29][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:29][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:30][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:30][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:30][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:31][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:31][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:31][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:31][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:31][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:31][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:32][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:32][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:32][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:32][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:32][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:32][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:33][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:33][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:33][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:33][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:33][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:34][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!
[08:23:34][D][homeassistant.sensor:024]: 'sensor.aq_temperature': Got state 22.48
[08:23:34][D][climate:396]: 'ac' - Sending state:
[08:23:34][D][climate:399]:   Mode: COOL
[08:23:34][D][climate:404]:   Fan Mode: AUTO
[08:23:34][D][climate:416]:   Swing Mode: VERTICAL
[08:23:34][D][climate:419]:   Current Temperature: 22.48°C
[08:23:34][D][climate:425]:   Target Temperature: 24.00°C
[08:23:34][D][sensor:103]: 'ac-Temp': Sending state 22.47629  with 1 decimals of accuracy
[08:23:34][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: RX: 00 A0 00 87 18 A4 01 20
[08:23:34][D][esphome::fujitsu_general_airstage_h_controller:103][UART_Event]: TX: 20 81 00 07 18 04 2D 00
[08:23:34][D][fujitsu_general::airstage::h::Controller:096][UART_Event]: UART break!